probe-rs/cargo-embed #5: Scrolling causes output to be unreadable

FrozenDroid, 2020-06-04:
**Describe the bug**
When using cargo embed rtt feature, once the output starts to become large enough to where the console starts scrolling, the output gets messed up.

**To Reproduce**
Steps to reproduce the behavior:
1. Print things until output doesn't fit vertically anymore
2. Be unable to read output due to missing characters all over the place

**Expected behavior**
No messed up output

**Desktop:**
 - NixOS 20.03
 - Alacritty 0.4.2

**Additional context**
Resizing the terminal fixes it temporarily.

Yatekii, 2020-06-04:
Hmm this sounds like something specific to your system :/ This seems to work fine to me.
I will investigate a little :)

Yatekii, 2020-06-24:
I cannot reproduce this :/ Can you maybe attach a screenshot and also try with another terminal emulator maybe? :)

mattiasgronlund, 2020-07-17:
Could this just be a bad terminal configuration.
It just sounds like the terminal has different size than what cargo embed thinks...

closed 2020-11-19 by Yatekii