probe-rs/cargo-embed #9: RTT long text output doesnt horitzontally wrap

jacobrosenthal, 2020-06-13:
If you print out a large datastructure or something it doesnt wrap the line at the column edge, nor are there scroll bars, to my knowledge, to see that theres data hidden. 

Note This isnt just cosmetic, Ive had a few rtt panics lost in the unviewable space as a result.

Were using the List type but it currently doesnt support wrapping
https://github.com/fdehau/tui-rs/issues/90

Paragraph does handle wrapping, but didnt appear to handle whatever newlines writeln! uses so it gave wrapping but hurt readability.

It apparently is supposed to handle some newlines?
https://github.com/fdehau/tui-rs/blob/4f728d363f0be75f844f39b2fa2705e7b4373953/examples/demo/ui.rs#L215

Yatekii, 2020-06-17:
Ah yeah, I guess I never tested this with wrapping lines. Pitty this doesn't work properly.
I wonder what we could do here :/ To implement scrolling properly, we would have to break the message into several lines ourselves. But that break would depend on the terminal width. An it would mess up the arangements on terminal resizing :/

How it works atm is that I fetch the data from RTT, look for \n's and split the string there. Then I insert all the strings into a list, which then gets rendered (that's how the TUI works). But if there is no newline and it starts wrapping, I am sure this fucks it up.

So I think clipping the contents to terminal width is not useful. Manually breaking the line is not so nice either ...
I guess we would need to fix the widget ourselves.

If anyone has a better idea, it is most welcome :)

jacobrosenthal, 2020-06-25:
BTW I noticed, paragraph DOES respect newlines. Its us who strip them out with split_terminator :) But again we dont have a way to do vertical scrolling with upstream paragraph nor can we do it locally because we dont know how many lines paragraph will be.

jacobrosenthal, 2020-07-16:
A naive solution was employed in #12, though long term we should push something native upstream and use their stateful widgets as described in #13

closed 2020-07-16 by jacobrosenthal

TheButlah, 2022-09-19:
Is this actually solved? I see this on my macbook:
[image]

If I expand the window:
[image]

I can't expand it far enough to actually see the content of the printed struct.

Yatekii, 2022-09-19:
Maybe it broke with a library update? I am pretty sure this was fixed. Or is it maybe also dependant on the OS?