probe-rs/cargo-embed #13: Implement scrolling

jacobrosenthal, 2020-06-24:
In looking into how to enable scrolling natively to tui-rs, as a part of #9 and #12 Ive been looking at how to implement scrolling lists, and presumably any other stateful widget types that tui might offer in the future. They have a new [render_stateful_widget](https://github.com/fdehau/tui-rs/blob/master/CHANGELOG.md#features-1) function that takes a &mut state and will highlight that item or scroll to it(in future) so we don't have to implement a lot of that ourselves.

If wed like to use that, Ive found our current architecture is kinda annoying due to interior immutability. 

We currently have an App that owns both terminal AND current_tab, where
current_tab would probably own the mut app state. Thus this fails to compile
self.terminal.draw(||
	f.render_stateful_widget(messages,chunks[1],&mut self.current_tab_mut().state)
)

Its not an accident in the examples that terminal and app are created top level
https://github.com/fdehau/tui-rs/blob/8c2ee0ed8516be887f5f68355c325fadecb5d2dc/examples/termion_demo.rs#L37-L42
https://github.com/fdehau/tui-rs/blob/2b48409cfd21ce91e00e2a45fdfe24b3c825ef69/examples/demo/ui.rs#L118

We could wrap current_tab_muts in refcell but that feels bad.
Thoughts

Yatekii, 2020-06-24:
Well, reaching everything in from top down is not feasible for any real application. You need to store the state somewhere. The problem here is that rust is not able (yet) to partially borrow in a method.
`self.current_tab_mut()` will fully borrow self. It should be easy to remodel it slightly such that this wont do a full borrow of self :)

jacobrosenthal, 2020-08-24:
Couple things in the pipeline that could clean up our terrible memory usage

https://github.com/fdehau/tui-rs/pull/349
https://github.com/fdehau/tui-rs/pull/361