probe-rs/cargo-embed #29: `.cargo/config.toml` is not recognized

yvt, 2020-07-17:
**Describe the bug**
`cargo-embed` fails to flash when `rustflags` is specified by `.cargo/config.toml` (instead of `.cargo/config`, which is more traditional).

**To Reproduce**
Steps to reproduce the behavior:
1. Clone <https://github.com/jkristell/nucleo-f401re>
2. Rename `.cargo/config` to `.cargo/config.toml`
3. `cargo embed --example timer-blinky --release`

**Expected behavior**
`cargo-embed` should successfully flash the program.

**Actual behavior**

```
       Error failed to flash /private/tmp/nucleo-f401re/target/release/examples/timer-blinky
```

**Desktop (please complete the following information):**
 - macOS Mojave

**Additional context**
Related issue: <https://github.com/japaric/cargo-project/issues/9>

Yatekii, 2020-07-17:
Thanks for reporting this. As you have recognized, this is a cargo-project issue. I will leave this open tho, until it has been fixed on the other side :)

benmkw, 2020-08-29:
actually I "fixed" it in cargo-project but turns out we are not even using it anymore since https://github.com/probe-rs/cargo-embed/commit/f48bd3b7be900de5c256fa3ca1477128c8004309#diff-80398c5faae3c069e4e6aa2ed11b28c0L17 so this can probably be closed the same as https://github.com/probe-rs/cargo-flash/issues/42#issuecomment-671543722 was :D 

@Tiwalun

Yatekii, 2020-08-29:
Fixed in #45 :)

closed 2020-08-29 by Yatekii