probe-rs/cargo-embed #34: `cargo embed` does not work, `cargo flash` does work; invalid preamble

jounathaen, 2020-08-09:
I have a project for an STM32F4 which I cannot flash with `cargo embed` due to the following error. **However `cargo flash` works fine.**

```
> cargo embed
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
    Flashing /xxx/target/thumbv7em-none-eabihf/debug/blinky
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
       Error failed attaching to target

Caused by:
    0: Unable to load specification for chip
    1: The connected chip could not automatically be determined.

> cargo flash --chip STM32F411CEUx --protocol swd
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
    Flashing /xxx/target/thumbv7em-none-eabihf/debug/blinky
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Breakpoint), new reason is Request
        WARN probe_rs::architecture::arm::core::m4 > Reason for halt has changed, old reason was Halted(Request), new reason is Exception
     Erasing sectors ✔ [00:00:00] [##################################]  32.00KB/ 32.00KB @  41.93KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [##################################]  21.00KB/ 21.00KB @  10.63KB/s (eta 0s )

```

Embed.toml:
```toml
[probe]
protocol = "swd"

[flashing]
enabled = true
halt_afterwards = false
restore_unwritten_bytes = false

[general]
chip = "STM32F411CEUx"
chip_descriptions = []
log_level = "Warn"
```

Both tools are v0.8.0 and installed via `cargo install` on Linux/Fedora32. I observed the same behaviour with STM32F1 chips, but I don't have any of theme at hand for reproduction

Yatekii, 2020-08-09:
Hmm this is very odd as they should both use the same probe-rs version and thus the same mechanism to perform the tasks required for flashing ...
But, I have to admit, I had divergent behavior before too.

We have to look into this, thanks for reporting :)

jack6000, 2020-08-15:
I have the same problem with a bluepill board.

However, replacing the section name [general] by [default.general] in Embed.toml works for me. Don't know why

Yatekii, 2020-08-15:
Oh, you are both using the very first version of the config format.

Default overrides need to be prefixed with `default.`

See: https://github.com/probe-rs/cargo-embed/blob/master/src/config/default.toml

and https://github.com/probe-rs/cargo-embed/blob/master/CHANGELOG.md#080

I hope that helps!

closed 2020-08-15 by Yatekii