probe-rs/cargo-embed #98: powershell gets confused after running embed

johnt7, 2020-10-12:
When running the Ferrous sample hello.rs in the beginners training with embed, the power shell line editing is confused afterwards.
Running Visual Studio code
Running a powershell terminal
run: cargo embed --bin hello  (in the beginner\apps directory)
I see the messages flashing etec, then the results, showing Hello, world.
I press Ctrl-C to stop it.
Now whenever I type in the powershell, it is not formatting properly.
Starting a new terminal solves the problem.

**Desktop (please complete the following information):**
 - Windows 10
latest updates 2020-10-11
visual studio code latest updates

sample code:
https://github.com/ferrous-systems/embedded-trainings-2020

johnt

Yatekii, 2020-11-08:
Hey @johnt7!

Is this still a problem? :) Could you try with current master?

johnt7, 2020-11-10:
So I tried setting up the sample from the knurling template:
https://github.com/knurling-rs/app-template

I updated with rustup,
Got the latest probe-run:
cargo install --git https://github.com/knurling-rs/probe-run --branch main --features defmt

and made sure cargo generate and flip-link were installed.

I generated a new project:
cargo generate --git https://github.com/knurling-rs/app-template --branch main --name co2test1<https://github.com/knurling-rs/app-template%20--branch%20main%20--name%20co2test1>

then updated the runner, target and memory layout.
When I ran it, I got the following.  Clearly I have something out of step, but not sure what to look at.

PS C:\projects\Rust\ferrous\co2test1> cargo run --bin hello
    Finished dev [optimized + debuginfo] target(s) in 0.10s
     Running `probe-run --chip nRF52840_xxAA --defmt target\thumbv7em-none-eabihf\debug\hello`
Error: defmt version mismatch (firmware is using 144184a6ac614146c45122ece7c9fc698b749719, host is using a6dc0904872cd3f4aee5125dffc17723df3c9393); are you using the same git version of defmt and related tools?
Try `cargo install`-ing the latest git version of `probe-run` AND updating your project's `Cargo.lock` with `cargo update`
error: process didn't exit successfully: `probe-run --chip nRF52840_xxAA --defmt target\thumbv7em-none-eabihf\debug\hello` (exit code:
1)
PS C:\projects\Rust\ferrous\co2test1>

Thanks
johnt

Yatekii, 2020-11-10:
I cannot help you with that unfortunately. I am not involved with knurling. Maybe you have better luck asking over there :)

closed 2020-11-10 by Yatekii

johnt7, 2020-11-10:
Thanks, I will do that

johnt