probe-rs/cargo-embed #99: New version 0.9.1 gives "Error Probe could not be created" on macOS Catalina

coolaj86, 2020-10-17:
**Describe the bug**

After upgrading to `cargo-flash 0.9.0` and `cargo-embed 0.9.1` on macOS I can no longer flash my STM32 Blue Pill. Instead I get this error message:

```txt
   Compiling blue_pill_base v0.1.0 (/Users/aj/BluePill_Rust/utah-rust-blue-pill-code)
    Finished release [optimized + debuginfo] target(s) in 5.29s
    Flashing /Users/aj/BluePill_Rust/utah-rust-blue-pill-code/target/thumbv7m-none-eabi/release/blue_pill_base
       Error Probe could not be created

Caused by:
    0: Pipe error
    1: Pipe error
```

Switching back to `cargo-flash 0.8.0` and `cargo-embed 0.8.0` solves this problem.

**To Reproduce**

 I'm working from these instructions https://cgit.pinealservo.com/BluePill_Rust/resources/issues/1, trying to update them for v0.9.x.

I have a Blue Pill STM32 plugged into my USB, with libusb drivers installed on macOS.

```bash
git clone git clone https://cgit.pinealservo.com/BluePill_Rust/blue_pill_base ./utah-rust-blue-pill-code/
pushd /utah-rust-blue-pill-code/
rustup run 1.46.0 \
	cargo install cargo-flash --version 0.9.0
rustup run 1.46.0 \
	cargo install cargo-embed --version 0.9.1
cargo embed --release
```

**Expected behavior**

I expect that v0.9.x should flash my STM32 as well or better than v0.8.x

**Desktop (please complete the following information):**
 - macOS Catalina

Yatekii, 2020-10-17:
Thanks for reporting this! It is a known issue (already reported in the probe-rs repo) but I have no way of debugging it as I do not own a mac.
The error must be within rusb or in the way we use it. Maybe thid is fixed on master. Could you try installing from cargo-embed master with probe-rs master? :)

coolaj86, 2020-10-17:
@Yatekii I'm still fairly new to the rust tooling ecosystem. How does one specify to install from master?

```bash
git clone https://github.com/probe-rs/cargo-embed
git push ./cargo-embed/
cargo ??? probes-rs --branch master
cargo install --path .
```

Yatekii, 2020-10-17:
Try `cargo install --git https://github.com/probe-rs/cargo-embed` :)

coolaj86, 2020-10-17:
```
cargo install --git https://github.com/probe-rs/cargo-embed
cargo embed --release

    Finished release [optimized + debuginfo] target(s) in 0.33s
      Config default
      Target /Users/aj/BluePill_Rust/utah-rust-blue-pill-code/target/thumbv7m-none-eabi/release/blue_pill_base
     Erasing sectors ✔ [00:00:00] [################################################################]   7.00KB/  7.00KB @  24.59KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [################################################################]   7.00KB/  7.00KB @   6.78KB/s (eta 0s )
    Finished flashing in 0.622s
       Error Error attaching to RTT: Error communicating with probe: An error with the usage of the probe occured
```

Yatekii, 2020-10-17:
Yay, progress!

I am not sure about the next one, there is parts of the message missing :)

Do you have some more info on what firmware you run?

coolaj86, 2020-10-17:
> Do you have some more info on what firmware you run?

I'm not clear on what you mean by `firmware` in this case.

I think of `firmware` as what I'm trying to flash to the microcontroller, but I don't think that's what you mean since `cargo embed 0.8.0` is what I've been using to flash the device with said firmware (which is this blinky light program: https://cgit.pinealservo.com/BluePill_Rust/blue_pill_base).

What command might I run to give you more details.

coolaj86, 2020-10-17:
In other news: `cargo-flash` from master _does_ work:

```bash
cargo install --git https://github.com/probe-rs/cargo-flash
cargo flash --release
```

```txt
    Finished release [optimized + debuginfo] target(s) in 0.03s
    Flashing /Users/aj/BluePill_Rust/utah-rust-blue-pill-code/target/thumbv7m-none-eabi/release/blue_pill_base
     Erasing sectors ✔ [00:00:00] [#######################]   7.00KB/  7.00KB @  31.02KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [#######################]   7.00KB/  7.00KB @   7.62KB/s (eta 0s )
```

Yatekii, 2020-10-17:
Well about the firmware question: If the firmware uses `wfi();` which RTIC applications with no idle() for example do, RTT wont work on most chips :) That's why I asked.

But the error does not look firmware related. You could try setting the `speed` property to 100. This will be slow but for experimentation purposes it could help :)

coolaj86, 2020-10-17:
@Yatekii I updated `Embed.toml`, but it didn't make a difference.

```yml
[default.probe]
# USB vendor ID
# usb_vid = "1337"
# USB product ID
# usb_pid = "1337"
# Serial number
# serial = "12345678"
# The protocol to be used for communicating with the target.
protocol = "Swd"
# The speed in kHz of the data link to the target.
speed = 100
```

Here's the full `Embed.toml` for the project (without the speed change):
https://cgit.pinealservo.com/BluePill_Rust/blue_pill_base/src/branch/master/Embed.toml

Any idea why `cargo-flash` works from master while `cargo-embed` does not? That seems like a clue.

Yatekii, 2020-10-17:
Both succeed to flash :) `cargo-embed` only fails to attach to RTT :) But without seeing the full message it's hard to tell :) Maybe it was just a fluke, maybe it's real trouble, I can't say :) Normally when flashing works RTT works too :)

Depending on your blue pill it might be a HW issue :)

coolaj86, 2020-10-17:
@Yatekii 

> without seeing the full message it's hard to tell

How can I print the full message? I don't see a `--verbose` option to `cargo embed`.

> Depending on your blue pill it might be a HW issue.

Why would it work with an older version just fine if it were hardware?

Luckily, I have a box full of these. I just tried two more. All 3 failed.

Yatekii, 2020-10-17:
> Why would it work with an older version just fine if it were hardware?

Good point! So RTT works withg older versions? Sorry I didn't account for that :/

> How can I print the full message? I don't see a `--verbose` option to `cargo embed`.

I think you omitted parts of the message stack printed, am I wrong? You can set the log level in the Embed.toml to DEBUG instead of WARN :)

Hmm it's odd that flashing would work but RTT fails :/ I guess the only way to really debug this might be bisecting commits :/ I am kinda poking into the dark here ... real hard to tell remotely :/

coolaj86, 2020-10-17:
Hmmm... unfortunately there's no more error output:

```txt
    Finished flashing in 4.226s
        INFO cargo_embed                            > Initializing RTT (attempt 1)...
       DEBUG probe_rs_rtt::rtt                      > Scanning at exact address: 0x20000000
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=20000000, length=24
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=20000018, length=24
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=0800177b, length=128
       DEBUG cargo_embed                            > Failed to initialize RTT. Retrying until timeout.
        INFO cargo_embed                            > Initializing RTT (attempt 2)...
       DEBUG probe_rs_rtt::rtt                      > Scanning at exact address: 0x20000000
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=20000000, length=24
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=20000018, length=24
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=0800177b, length=128
       DEBUG cargo_embed                            > Failed to initialize RTT. Retrying until timeout.
        INFO cargo_embed                            > Initializing RTT (attempt 3)...
       DEBUG probe_rs_rtt::rtt                      > Scanning at exact address: 0x20000000
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=20000000, length=24
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=20000018, length=24
       DEBUG probe_rs::probe::stlink                > Read mem 8 bit, address=0800177b, length=128
       DEBUG cargo_embed                            > Failed to initialize RTT. Retrying until timeout.
       DEBUG probe_rs::probe::stlink                > Read mem 32 bit, address=e0002000, length=1
       DEBUG probe_rs::probe::stlink                > Current device mode: Jtag
       Error Error attaching to RTT: Error communicating with probe: An error with the usage of the probe occured
```

> real hard to tell remotely

If you'd care to give me an SSH key, I'll be happy to let you in to <stm32@stm32.duckdns.org>. :)

Yatekii, 2020-10-17:
Sure :) Maybe not tonight anymore tho :D Maybe you can also join the Matrix channel for easier comms :)

coolaj86, 2020-10-24:
@Yatekii Could you cut a new release?

The latest master is working again:

```bash
cargo install --force --git https://github.com/probe-rs/cargo-embed
```

I can now see debug messages in the terminal. :D

Yatekii, 2020-10-24:
Yeah, I think we should be able to make a release :)

coolaj86, 2020-10-30:
@Yatekii ETA on the release?

Yatekii, 2020-10-30:
I don't know. We are still battling some issues.

Yatekii, 2020-11-08:
Afaik this issue is resolved and the fix will be part of a iminent 0.10.0 release. Please reopen if this is still an issue :)

closed 2020-11-08 by Yatekii