probe-rs/cargo-embed #119: cargo embed fails to build virtual manifest

berkowski, 2020-11-02:
**Describe the bug**
`cargo embed` fails to build virtual manifest crates.  A standard `cargo build` builds successfully builds the target. 

**To Reproduce**
An example virtual manifest crate is here: [cargo-embed-workspace](https://github.com/berkowski/cargo-embed-workspace).  It is set up to build for stm32l0x1 but can be easily adapted to another flavor for local testing.  Running `cargo build` in the crate root builds the firmware target as desired, `cargo embed` fails on a dependency check.

**Expected behavior**
`cargo embed` should at least be able to build the firmware exactly as `cargo build`

**Stacktrace**
```
$ cargo embed
error: package collision in the lockfile: packages common v0.1.0 (C:\Users\Zac Berkowitz\Documents\example\common) and common v0.1.0 (C:\Users\Zac Berkowitz\Documents\example\firmware\../common) are different, but 
only one can be written to lockfile unambiguously
       Error Failed to run cargo build: exit code = Some(101)
```

**Desktop (please complete the following information):**
 - Windows 10
 - Ubuntu 18.04

**Additional context**
 The same code worked with `cargo embed` before when the packages were not part of a common workspace.  The workspace is used to split device-agnostic code out and enable some amount of unit-testing on the host machine.  Initially the code didn't use workspaces due to [cargo-#7004](https://github.com/rust-lang/cargo/issues/7004).

Tiwalun, 2020-11-03:
Unfortunately I'm not able to reproduce this with the example you provided.

When running `cargo embed` in the root directory, it seems to work as expected:

```
❯ cargo embed
    Updating crates.io index
    Finished dev [optimized + debuginfo] target(s) in 1.08s
      Config default
      Target /home/dominik/Coding/cargo-embed-workspace/target/thumbv6m-none-eabi/debug/firmware
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR1 has invalid preamble (expected 0x0, got 0x1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR2 has invalid preamble (expected 0x5, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
       Error failed attaching to target
             
             Caused by:
                 0: Unable to load specification for chip
                 1: The connected chip could not automatically be determined.

```
This was on Ubuntu 20.10. 

What version of `cargo embed` are you using?

berkowski, 2020-11-03:
Looks like it's a windows thing.  I tried again with Ubuntu 18.04 and it compiles OK.  Must not have had a clean environment.

Still fails on windows though with cargo-embed 0.9.1

Tiwalun, 2020-11-05:
This seems to be due to https://github.com/rust-lang/cargo/issues/6198, we do normalize the path before running Cargo.

Tiwalun, 2020-11-05:
https://lib.rs/crates/dunce seems to be a useful alternative to the built-in `std::fs::canonicalize`, which hopefully avoids the whole mess with UNC paths.

berkowski, 2020-11-05:
Ahh, yes UNC paths.  They've bitten me a few times already.  If you point me towards a starting point I can help with a PR.

Tiwalun, 2020-11-05:
I think replacing the call to `canonicalize in [this file](https://github.com/probe-rs/probe-rs/blob/7b33b4b42ea1de33c98cb2e46986217e7c86f515/probe-rs-cli-util/src/lib.rs#L45) in probe-rs
with the equivalent function in the dunce crate should be enough to fix this issue.

berkowski, 2020-11-05:
Should be closed with probe-rs/probe-rs#416

closed 2020-11-05 by berkowski