probe-rs/cargo-embed #145: cargo-embed wants to send Sentry report, if build fails

hannobraun, 2020-11-20:
**Describe the bug**
See title.

**To Reproduce**
Steps to reproduce the behavior:
1. Introduce a build error into a project that uses cargo-embed.
2. Run `cargo-embed`
3. Observe the prompt asking to send a report to Sentry.

**Expected behavior**
I would expect that this prompt only appears, if some unexpected error within cargo-embed occurs.

**Stacktrace**
Not a stack trace, but here's the output:
```
$ cargo embed
   Compiling stm32l4-test-target v0.1.0 (/home/hanno/Projects/braun-embedded/embedded-test-stand/stm32l4-test-stand/test-target)
error: could not compile `stm32l4-test-target`.

To learn more, run the command again with --verbose.
error[E0283]: type annotations needed for `stm32l4xx_hal::dma::FrameSender<BUFFER, stm32l4xx_hal::dma::dma1::C4, N>`
   --> src/main.rs:172:52
    |
172 |                         let frame_sender = tx_main.frame_sender(dma1::C4);
    |                             ------------           ^^^^^^^^^^^^ cannot infer type for type parameter `BUFFER` declared on the associated function `frame_sender`
    |                             |
    |                             consider giving `frame_sender` the explicit type `stm32l4xx_hal::dma::FrameSender<BUFFER, stm32l4xx_hal::dma::dma1::C4, N>`, where the type parameter `BUFFER` is specified
    |
    = note: cannot satisfy `_: stable_deref_trait::StableDeref`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0283`.
       Error Failed to run cargo build: exit code = Some(101)
        Hint Unfortunately probe-rs encountered an unhandled problem. To help the devs, you can automatically log the error to sentry.technokrat.ch. Your data will be transmitted completely anonymous and cannot be associated with you
             directly. To hide this message in the future, please set $PROBE_RS_SENTRY to 'true' or 'false'. Do you wish to transmit the data? Y/n: 
```

**Desktop (please complete the following information):**
Arch Linux

Yatekii, 2020-11-20:
Ohh this is real bad :/

I hope it's okay for you to just permanently disable/enable it via environment variable :)

I will try to deploy an immediate fix tonight!

hannobraun, 2020-11-20:
Not a big deal for me, I'm just running `cargo build` more often as a workaround. Thanks for looking into it!

Yatekii, 2020-12-16:
Sentry is disbled by default for now :)

closed 2020-12-16 by Yatekii