probe-rs/cargo-embed #167: Fix UX paper-cut - invalid fields (e.g. typos) in configuration files are silently ignored

tim-seoss, 2020-12-14:
It's currently possible to add fields to the configuration sources which are ignored (e.g. because they're completely invalid, or typos).  e.g.

```toml
[default.gdb]
# This typo (should be "enabled", NOT "enable") should be flagged, but it's silently ignored
enable = true
```

I've dug about in the config code, and it looks non-trivial to add this feature because of the way nesting has been implemented atop `config`.  One possible solution would be to switch from using the `config` crate (which seems to be defacto deprecated by its maintainer) to `figment` - https://crates.io/crates/figment - which can be made to trivially detect this, and report the source (e.g. file path, and line number) of the spurious entry.

Yatekii, 2020-12-14:
Yeah I kinda expected it to be troublesome, because of the config crate. Figment sounds like a a nice solution to me :)

tim-seoss, 2020-12-14:
The only possible downside I can see to `figment` is that it will summon profiles into existence if they are requested, but not defined.  I'll look into it (which in turn would allow typo'ing profiles instead of field names).  I'll look into what can be done about that, but it might be sufficient to warn that the requested profile == the default profile.

tim-seoss, 2020-12-14:
Looks like that functionality (enumerate all defined profiles) is planned for figment.  I'll have a go at incorporating figment if I get a spare minute (the next week is busy, so probably after that).

tim-seoss, 2020-12-16:
n.b. Figment added profile enumeration with: https://github.com/SergioBenitez/Figment/commit/d152c2a236496b0dafe74e6cc5540c7d166a94f2

Yatekii, 2020-12-16:
Very nice! So were basically waiting on a release :)

Tiwalun, 2021-02-01:
This seems to be released now: https://docs.rs/figment/0.10.2/figment/struct.Figment.html#method.profiles

@tim-seoss Are you still planning to work on this?

tim-seoss, 2021-02-04:
@Tiwalun  Still planning, still busy.  Will try and get onto it next week...

commit e0a1767c (2021-04-28): Switch to the figment crate for configuration parsing.

The `config` crate has some limitations which make it undesirable.
`figment` has improved error reporting, and allows the user to be
alerted if they define invalid (i.e. unknown) configuration items.

Closes #167.
closed 2021-04-28 by bors[bot]