probe-rs/cargo-embed #168: Build fails due to probe-rs renames

dbrgn, 2020-12-15:
See https://github.com/probe-rs/probe-rs/pull/482.

Yatekii, 2020-12-15:
This is fixed already and is not cargo-embed related but probe-rs-rtt :)

noppej, 2020-12-15:
@Yatekii Thank you for fixing this. I have a 'newbie' question ... can you please give me some pointers on how I can install cargo-embed with this new version of probe-rs-rtt?

Yatekii, 2020-12-15:
@noppej you should be able to install with `cargo install --git https://github.com/probe-rs/cargo-embed`. Let me know if this works :)

noppej, 2020-12-16:
Thanks, but it fails with the same error ... I don't know how to make it pick up the right version of probe-rs-rtt.

```
   Compiling probe-rs-rtt v0.10.1 (https://github.com/probe-rs/probe-rs-rtt#b06d69d7)
error[E0599]: no variant or associated item named `Nvm` found for enum `probe_rs::config::MemoryRegion` in the current scope
   --> /Users/Jack/.cargo/git/checkouts/probe-rs-rtt-7666cc4c60dc073f/b06d69d/probe-rs-rtt/src/channel.rs:383:27
    |
383 |             MemoryRegion::Nvm(r) => Some(&r.range),
    |                           ^^^ variant or associated item not found in `probe_rs::config::MemoryRegion`

```

Yatekii, 2020-12-16:
So this failed because patch versions are ignored when you install with --git apparently. if you try again, this should hopefully work :)

dbrgn, 2020-12-16:
> This is fixed already and is not cargo-embed related but probe-rs-rtt :)

Whoops, sorry! :slightly_smiling_face:

dbrgn, 2020-12-16:
@noppej maybe you also need a `cargo update`.

noppej, 2020-12-16:
Thank you @Yatekii ... the patch thing fixed. Much appreciated!

closed 2020-12-16 by Yatekii