probe-rs/cargo-embed #169: Is there a way to direct RTT to stdout rather than rttui?

noppej, 2020-12-17:
Hi. I'm trying to integrate cargo-embed (with both RTT and GDB) into a VSCode debug extension. Without the RTT output, it behaves rather well, but as soon as I turn on RTT, it causes a problem.   

The rttui/app.rs makes calls to `crossterm::enable_raw_mode()`, which causes a panic  
` code: 102, kind: Other, message: "Operation not supported on socket"`

Is there a config (or other) way will allow me to bypass rttui, and have rtt output go directly to `stdout`?

Yatekii, 2020-12-17:
Currently there is no such way. I have a complete overhaul in mind which should solve this going into the future.
If you want this right now, I think you'd have to change it in the code yourself :) I'll accept PRs doing this :)

One idea I had was opening one socket per RTT channel and using something like `socat`/`netcat` to operate it. This would fix many issues.

noppej, 2020-12-17:
OK ... let me clean up the debugger integration (gdb minus RTT)  and then I'll give this some thought. Ping me if you have any other suggestions or things you want me to work on while I'm in that part of the code. Also, are you OK if I ping you here if I have questions or want to bounce ideas around?

Yatekii, 2020-12-17:
Sure, ping whenever & how often you want, I am not bothered much by it and I like chatting :P

You could also join matrix to have a more interactive chat :)

Also, are you aware of the https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug extension? :) It does not support RTT of course tho.

Also, are you aware of our GDB free extension: https://github.com/probe-rs/vscode it can currently display a stacktrace and the names of local variables I think. I think it can also step & run & halt. Breakpoints are a construction site still I believe. So if you are interested, it would definitely be a nice thing to continue work here as GDB really is not the future :)

noppej, 2020-12-17:
Cool. I have used the cortex-debug extension a lot with OpenOCD and GDB (with c language projects). I'm actively trying to add a config to it for connecting to cargo-embed, and that is why I ran into the rtt problems. I really like the cortex-debug because of the easy SVD peripheral browser and general stability/support.

I looked at the probe-rs/vscode extension but it felt like it has a long way to go and the cortex-debug extension has lots of history and active development. Are you OK if I continue down the path of the cortex-debug?

Yatekii, 2020-12-18:
Yeah I mean if RTT makes troubles and you don't need it, you can just deactivate it and then cortex-debug should work fine :)

Of course, I am okay with whichever you like, it's your time :) Ofc I would prefer the dedicated plugin, as possibilities are basically limitless, but it doesn't have to be now :) If you are after a quick solution, GDB based is the way :)