probe-rs/cargo-embed #189: cargo install cargo-embed fails

wez, 2021-02-14:
**Describe the bug**

`cargo install cargo-embed` fails to compile:

```
error[E0034]: multiple applicable items in scope
   --> /home/wez/.cargo/registry/src/github.com-1ecc6299db9ec823/bitvec-0.19.4/src/field.rs:307:25
    |
307 |         check("load", len, M::BITS);
    |                               ^^^^ multiple `BITS` found
    |
note: candidate #1 is defined in the trait `BitMemory`
   --> /home/wez/.cargo/registry/src/github.com-1ecc6299db9ec823/bitvec-0.19.4/src/mem.rs:29:2
    |
29  |     const BITS: u8 = mem::size_of::<Self>() as u8 * 8;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: candidate #2 is defined in the trait `IsNumber`
   --> /home/wez/.cargo/registry/src/github.com-1ecc6299db9ec823/funty-1.2.0/src/lib.rs:144:2
    |
144 |     const BITS: u32;
    |     ^^^^^^^^^^^^^^^^
help: disambiguate the associated constant for candidate #1
    |
307 |         check("load", len, BitMemory::BITS);
    |                            ^^^^^^^^^^^^^^^
help: disambiguate the associated constant for candidate #2
    |
307 |         check("load", len, IsNumber::BITS);
    |                            ^^^^^^^^^^^^^^
```

This is on Fedora 33, amd64.

Installing from the git repo doesn't succeed either:

```
$ cargo install cargo-embed --git https://github.com/probe-rs/cargo-embed
   Compiling cargo-embed v0.10.1 (/home/wez/.cargo/git/checkouts/cargo-embed-7664de7e4a0e8cc2/14e8d44)
error[E0063]: missing field `do_chip_erase` in initializer of `DownloadOptions<'_>`
   --> src/main.rs:434:17
    |
434 |                 DownloadOptions {
    |                 ^^^^^^^^^^^^^^^ missing `do_chip_erase`

error[E0063]: missing field `do_chip_erase` in initializer of `DownloadOptions<'_>`
   --> src/main.rs:452:17
    |
452 |                 DownloadOptions {
    |                 ^^^^^^^^^^^^^^^ missing `do_chip_erase`

error: aborting due to 2 previous errors
```

wez, 2021-02-14:
Checking out the git repo; `cargo build --release` succeeds but `cargo install --path .` fails.
Removing the `Cargo.lock` causes `cargo build --release` to also fail; it seems that something in `Cargo.toml` is under-specified?

wez, 2021-02-14:
This patch makes `cargo install --path .` succeed for me:

```diff
diff --git a/src/main.rs b/src/main.rs
index 0149d16..7206ad2 100644
--- a/src/main.rs
+++ b/src/main.rs
@@ -434,6 +434,7 @@ fn main_try() -> Result<()> {
                 DownloadOptions {
                     progress: Some(&progress),
                     keep_unwritten_bytes: config.flashing.restore_unwritten_bytes,
+                    do_chip_erase: false,
                 },
             )
             .with_context(|| format!("failed to flash {}", path.display()))?;
@@ -452,6 +453,7 @@ fn main_try() -> Result<()> {
                 DownloadOptions {
                     progress: None,
                     keep_unwritten_bytes: config.flashing.restore_unwritten_bytes,
+                    do_chip_erase: false,
                 },
             )
             .with_context(|| format!("failed to flash {}", path.display()))?;
```

Yatekii, 2021-02-14:
So there is two things here. A dependency of a dependency, namely `funty` apparently broke semver rules which causes this. `probe-rs` master already has a fix merged. But we did some overhaul of probe-rs erroring which requires some adaptations of `cargo-flash/embed` which is a hen-egg problem :/ cargo-flash already has PR open that is +- ready to merge but cargo-embed doesn't.

In the meantime, I believe you should be able to make it work with `cargo install cargo-embed --locked` or maybe it is `--frozen` I am never sure.

commit 543a15cd (2021-03-10): Merge #190

190: Adapt to the flash api changes r=Yatekii a=Yatekii

Fixes #189 & #191.
PR #190 (2021-03-10): Adapt to the flash api changes
Fixes #189 & #191.
closed 2021-03-10 by bors[bot]