probe-rs/cargo-embed #191: Unable to use Defmt logging

seanybaggins, 2021-02-18:
Unable to use the Defmt logging.

Steps to reproduce the behavior:
My `Embed.toml`
```toml
[default.general]
chip = "STM32F303VCTx"

[default.rtt]
enable = true

channels = [
    { up = 0, name = "defmt_rtt", format = "Defmt" },
]
```

`cargo embed --version`
```
cargo-embed 0.10.1
git commit: crates.io
```
Error message from running `cargo embed`
```
       Error The config 'default' could not be loaded.
             
             Caused by:
                 enum DataFormat does not have variant constructor Defmt
```
Looking in my `~/.cargo/registry/src/github.com-xxxxxxxxxxx/cargo-embed-0.10.1/src/rttui/channel.rs` file.
```rust
#[derive(Debug, Copy, Clone, serde::Serialize, serde::Deserialize)]
pub enum DataFormat {
    String,
    BinaryLE,
}
```

This seems to contradict the [change log](https://github.com/probe-rs/cargo-embed/releases/tag/v0.9.1), [default.toml](https://github.com/probe-rs/cargo-embed/blob/master/src/config/default.toml) and [the source code in the master branch](https://github.com/probe-rs/cargo-embed/blob/master/src/rttui/channel.rs#L6-L11). It should be
```rust
#[derive(Debug, Copy, Clone, PartialEq, serde::Serialize, serde::Deserialize)]
pub enum DataFormat {
    String,
    BinaryLE,
    Defmt,
}
```  

I tried to install `cargo embed` using 
```
cargo install --git https://github.com/probe-rs/cargo-embed.git --branch master
```
but ran into issue #189.

Yatekii, 2021-02-18:
Sorry for the inconvenience! 0.10 does not have defmt logging :)

Building against probe-rs master is fixed with this PR: https://github.com/probe-rs/cargo-embed/pull/190 I hope to merge it today. You can try that branch :)

seanybaggins, 2021-02-22:
So I installed against the branch you suggested. I was able to install but now I am unable to launch a rtt terminal.

Using `cargo embed` I get 
```
Config default
      Target ${SOME_DIR}/ipendulum/implementation/target/thumbv7em-none-eabihf/debug/implementation
     Erasing sectors ✔ [00:00:02] [####################################################################]  46.00KB/ 46.00KB @  22.89KB/s (eta 0s )
 Programming pages   ✔ [00:00:04] [####################################################################]  45.00KB/ 45.00KB @   7.10KB/s (eta 0s )
    Finished flashing in 6.225s
        Done processing config default
```
and then I get put into my regular shell.

Here is my `Embed.toml`
```toml
 [default.general]
chip = "STM32F303VCTx"

[default.rtt]
enable = true

channels = [
    { up = 0, name = "defmt_rtt", format = "Defmt" },
]
```

Yatekii, 2021-02-24:
Does it just hang here? I hope I can test it n the evening :)