probe-rs/cargo-embed #199: Using cargo embed causes Windows terminals to behave strangely

AlyoshaVasilieva, 2021-03-03:
**Describe the bug**
Running cargo embed (e.g. `cargo embed --release --example crc --features stm32h743v,rt,log-rtt`) causes Windows terminals to behave strangely afterwards. Scrolling is no longer possible (anything that goes off the top of the console is gone forever) and the cursor is gone.

This happens with PowerShell and `cmd`. Windows Terminal appears to still be able to scroll but cursor is still gone.

**To Reproduce**
Steps to reproduce the behavior:
1. Run cargo embed in PowerShell or cmd
2. End with Ctrl-C

**Expected behavior**
Once cargo embed is ended terminal behaves normally, with scrolling and a cursor.

**Desktop (please complete the following information):**
Windows 10 20H2

Yatekii, 2021-03-04:
Hey, I assume you use RTT. You can scroll with pgup and pgdown. and with the F-Keys you can switch the tab.  characters & Enter can send something to a `down` channel.

AlyoshaVasilieva, 2021-03-04:
After terminating cargo embed (which was indeed using RTT) PowerShell/cmd cannot scroll and displays no cursor. I am no longer using RTT or cargo embed at that point; I have to exit and open a new terminal in order to have a terminal that behaves normally. PgUp/PgDn do not do anything here.

Yatekii, 2021-03-04:
Oh, I see. Is this 0.10.x or https://github.com/probe-rs/cargo-embed/tree/master-fix?

We had this problem a while back but we have fixed it. So it's bad it's happening again. That being said, depending on what shell you use on windows, this is really hard to fix.

AlyoshaVasilieva, 2021-03-04:
Tried master-fix branch, no change. Was on 0.10.1. However I think my repro steps were slightly wrong. While looking through the code I tried adding a `panic!` to the Ctrl-C handling code, which never triggered. This caused me to notice that I am pressing Ctrl-C twice: the first one does nothing (even if waiting a long time), the second exits cargo-embed.

I changed the beginning of `handle_event` to:

```rust
        let next = self.events.next().unwrap();
        println!("event: {:?}", next);
        match next {
```

(and derived Debug on Event) and I see this:

```
event: Tick
event: Tick
event: Tick
event: Tick
event: Tick
event: Tick
event: Tick
event: Tick
event: Tick
event: Input(KeyEvent { code: Char('c'), modifiers: CONTROL })
```

Nothing ever happens after the first Ctrl-C event, it seems to hang. Pressing Ctrl-C again somehow causes cargo-embed to exit, I think without running cleanup.

I'll see if I can stumble into a way to fix this, otherwise I'll just dedicate one terminal to running cargo embed and use another for doing other things.

jonnyando, 2021-04-22:
I am also seeing this behaviour on Windows 10 in both powershell and cmd, on version 0.10.1 from crates.io
Ctr+C freezes the UI, so if rtt messages are being printed they pause. Pressing Ctrl+C again kills the program.
After the program is killed the cursor does not return in either shell, and no log file is created/saved.

Tiwalun, 2021-11-05:
I've tried to reproduce this, but was not able to. RTT seemed to work normally, both in Powershell as well as in cmd.exe

Could you please try again to reproduce this with the current master branch?