probe-rs/cargo-embed #204: connect_under_reset from config file not respected

kuon, 2021-03-28:
If I put `connect_under_reset = true` under `[default.general]` in my `Embed.toml` file, it will not be picked up by `cargo-embed` or `cargo-flash`, I am forced to add `--connect-under-reset` to the command line for it to be applied. (I have version 0.10.1)

Yatekii, 2021-03-28:
You need to use the version from master (`cargo install --git https://github.com/probe-rs/cargo-embed`) where this is fixed. Sorry for the inconvenience!

kuon, 2021-03-29:
It seems to be fixed for `cargo-embed`, but not `cargo-flash`. Is it normal that cargo-flash ignores `Embed.toml` ? I use `cargo-flash` when I don't want an RTT console attached. One solution would be to add `rtt` flag to `cargo-embed` so I can use it in place of `cargo-flash` when I don't want an RTT console, without modifying the conf file.

Yatekii, 2021-03-29:
Cargo flash does not use the Embed.toml. That is to be expected, yes.

You can specify different profiles in the Embed.toml.
The default one is the one you use at the moment. You can create a new profile by specifying a different prefix to each config instead of `default`. You can use `default.rtt.enabled = false` and `rtt.rtt.enabled = true` for example and then call `cargo embed` for no RTT and `cargo embed rtt` to use the rtt profile. All profiles will inherit unset config flags from the default profile :)

closed 2021-03-29 by Yatekii

Yatekii, 2021-03-30:
No worries :) Well they are non-competing solutions. cargo-flash is just for quickly flashing something whereas cargo-embed is meant for debugging :)