probe-rs/cargo-embed #207: When RTT is cut, cargo-embed cannot be stopped with CTRL+C

kuon, 2021-03-30:
I have a firmware that I go into STOP mode under some conditions. I am testing those and sending it into STOP mode when connected to the debugger.

When this happen while `cargo-embed` is connected via RTT, the terminal is flooded (look at attached image). And except killing cargo from another terminal, there is no way to stop it.

It would be helpful if I could just CTRL+C to stop it when it's flooding. And it would even be better to stop the flood altogether, and just have a single log line that the connection was interrupted.

[image]

Yatekii, 2021-03-30:
Please provide more info about your enviroment. Ctrl+C works fine for me.

Yatekii, 2021-03-30:
Very odd.
Maybe my memory is bad then :/

willemml, 2021-05-18:
I am having the same issue on 0.10.1.
I am using macOS 11.2 and an STLink V3SET with an STM32F730V8T6.

Tiwalun, 2021-05-18:
I was able to reproduce this, on Ubuntu 21.04.

Probe: ST-Link v2
Terminals: gnome-terminal, alacritty
Chip: stm32f412zg

The problem seems to occur when the WFI instruction is used. See also probe-rs/probe-rs#350. Just unplugging the probe does not seem to cause this, in that case CTRL-C works.

commit 52b7eebf (2021-05-18): Improve input event handling (fix #207)

If the access to the probe takes a long time, input events are not
processed properly anymore. It seems that the tick events occur
too often in that case, and the actual input is not processed anymore.

By using recv_timeout, we do not need the Tick events, which seems
to solve this issue.
commit f2236355 (2021-05-18): Merge #222

222: Improve input event handling (fix #207) r=Yatekii a=Tiwalun

If the access to the probe takes a long time, input events are not
processed properly anymore. It seems that the tick events occur
too often in that case, and the actual input is not processed anymore.

By using recv_timeout, we do not need the Tick events, which seems
to solve this issue.
closed 2021-05-18 by bors[bot]