probe-rs/cargo-embed #213: Error when trying to `load` in gdb

chmanie, 2021-05-04:
Disclaimer: this is the first time I'm attempting to use cargo-embed. If I'm missing something obvious, please bear with me.

**Describe the bug**
I can't seem to be able to load a file using `gdb` when connected to the the integrated `gdb` stub server of cargo-embed. I'm getting the following error:

```
Reading symbols from target/thumbv7em-none-eabi/debug/blinky...
(gdb) target remote :1337
Remote debugging using :1337
0x00000000 in ?? ()
(gdb) load
Loading section .vector_table, size 0x400 lma 0x8000000
Load failed
```

<details>
  <summary>cargo-embed output</summary>
  
```
   Compiling stm32f446re v0.1.0 (/Users/chris/dev/rust/stm/stm32f446re)
    Finished release [optimized + debuginfo] target(s) in 37.02s
      Config default
      Target /Users/chris/dev/rust/stm/stm32f446re/target/thumbv7em-none-eabi/release/blinky
     Erasing sectors ✔ [00:00:00] [##########################################################################################]  16.00KB/ 16.00KB @  41.92KB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [##########################################################################################]   2.00KB/  2.00KB @   3.05KB/s (eta 0s )
    Finished flashing in 0.458s
Firing up GDB stub at localhost:1337.
GDB stub listening on localhost:1337
Accepted a new connection from: [::1]:55191
        WARN probe_rs_gdb_server::worker > WORKING qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+
        WARN probe_rs_gdb_server::worker > WORKING vMustReplyEmpty
        WARN probe_rs_gdb_server::worker > Unknown command: 'V(Unknown([77, 117, 115, 116, 82, 101, 112, 108, 121, 69, 109, 112, 116, 121]))'
        WARN probe_rs_gdb_server::worker > WORKING Hg0
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[72, 103, 48]': Parsing Error: Error { input: [72, 103, 48], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING qTStatus
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[113, 84, 83, 116, 97, 116, 117, 115]': Parsing Error: Error { input: [113, 84, 83, 116, 97, 116, 117, 115], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING ?
        WARN probe_rs_gdb_server::worker > WORKING qfThreadInfo
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[113, 102, 84, 104, 114, 101, 97, 100, 73, 110, 102, 111]': Parsing Error: Error { input: [113, 102, 84, 104, 114, 101, 97, 100, 73, 110, 102, 111], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING qL1200000000000000000
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[113, 76, 49, 50, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48]': Parsing Error: Error { input: [113, 76, 49, 50, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING Hc-1
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[72, 99, 45, 49]': Parsing Error: Error { input: [72, 99, 45, 49], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING qC
        WARN probe_rs_gdb_server::worker > Unknown command: 'Query(ThreadId)'
        WARN probe_rs_gdb_server::worker > WORKING qAttached
        WARN probe_rs_gdb_server::worker > WORKING qOffsets
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[113, 79, 102, 102, 115, 101, 116, 115]': Parsing Error: Error { input: [113, 79, 102, 102, 115, 101, 116, 115], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING Hg0
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[72, 103, 48]': Parsing Error: Error { input: [72, 103, 48], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING g
        WARN probe_rs_gdb_server::worker > WORKING g
        WARN probe_rs_gdb_server::worker > WORKING pf
        WARN probe_rs_gdb_server::worker > WORKING qL1200000000000000000
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[113, 76, 49, 50, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48]': Parsing Error: Error { input: [113, 76, 49, 50, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48, 48], code: Char }
        WARN probe_rs_gdb_server::worker > WORKING qXfer:memory-map:read::0,1000
        WARN probe_rs_gdb_server::worker > WORKING pd
        WARN probe_rs_gdb_server::worker > WORKING qSymbol::
        WARN probe_rs_gdb_server::worker > Failed to parse packet '[113, 83, 121, 109, 98, 111, 108, 58, 58]': Parsing Error: Error { input: [113, 83, 121, 109, 98, 111, 108, 58, 58], code: Char }
```
</details>

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new project using the stm32f4-hal, e.g. [delay-blinky](https://github.com/stm32-rs/stm32f4xx-hal/blob/bb160bbe8604bccf28557975131d0afdb2fa1014/examples/delay-blinky.rs)
2. Enable gdb in Embed.toml
3. Do `cargo embed --release --chip STM32F446RETx`
4. Run `arm-none-eabi-gdb -q target/thumbv7em-none-eabi/debug/delay-blinky`
5. Execute `target remote :1337` and `load`

My `Embed.toml` file:

```
[default.gdb]
enabled = true
```

**Expected behavior**
Here's the output when using `openocd` instead (`openocd -f interface/stlink-v2-1.cfg -f target/stm32f4x.cfg`):

```
❯ arm-none-eabi-gdb -q target/thumbv7em-none-eabi/debug/blinky
Reading symbols from target/thumbv7em-none-eabi/debug/blinky...
(gdb) target remote :3333
Remote debugging using :3333
0x08000610 in vcell::VolatileCell<u32>::set<u32> (self=0x0, value=0) at /Users/chris/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.3/src/lib.rs:41
41              unsafe { ptr::write_volatile(self.value.get(), value) }
(gdb) load
Loading section .vector_table, size 0x400 lma 0x8000000
Loading section .text, size 0x34d4 lma 0x8000400
Loading section .rodata, size 0x8f8 lma 0x80038d4
Start address 0x08000400, load size 16844
Transfer rate: 16 KB/sec, 5614 bytes/write.
```

**Desktop (please complete the following information):**
 - macOS Big Sur

**Additional context**
 - NUCLEO-F446RE board

Yatekii, 2021-05-04:
Hi!

`load` is currently not implemented. We only implemented GDB for use until we have better options. It's not something we want to support. That's why it does not have many features reaching over the basics.

You are free to implement it tho :) It should be trivial.

Tiwalun, 2021-05-04:
The recommended way of flashing with cargo-embed is to enable it in the config file:

```
[default.flashing]
# Whether or not the target should be flashed.
enabled = true
```

With this, the file is first flashed, and the GDB server is started afterwards. This should have the same behaviour as using `load` in GDB.

chmanie, 2021-05-04:
@Yatekii @Tiwalun thanks for helping out! 

> until we have better options

Did you agree on a way forward here? What's the most promising option here at the moment?

closed 2021-05-04 by chmanie

Yatekii, 2021-05-04:
Oh, sorry, I did not understand you did not flash at all. My answer is a bit lackluster ...

@better options:
There is an extensive VSCode plugin for it's debugger in the making with alpha releasing soon :)

chmanie, 2021-05-04:
Hooray! Is it this one? https://github.com/probe-rs/vscode

Maybe I can help with the implementation and/or testing.

Yatekii, 2021-05-04:
> Hooray! Is it this one? https://github.com/probe-rs/vscode
> 
> Maybe I can help with the implementation and/or testing.

This was the first start. @noppej is working on this and he'll publish the progress soon. I'll tag you when it goes public :)