probe-rs/cargo-embed #258: Rtt should be put in blocking mode

derekdreery, 2021-09-09:
The RTT on the device is started in non-blocking mode, to avoid locking the MCU if a probe is not connected. `cargo-embed` should but the device in non-blocking mode once the connection is established, to avoid losing data & corrupting the channel.[Here is how `probe-run` does it](https://github.com/knurling-rs/probe-run/blob/main/src/main.rs#L153)

Yatekii, 2021-09-09:
Well, it's a feature not a bug :) I do not care about 2-3 lost bytes in RTT, but I do care if $protocol misses a timing because RTT stalls :) So for probe-run it might be okay to do it this opinionated way, but I do not agree that this should be the default for cargo-embed. We could add a flag that makes it blocking on join tho :)

derekdreery, 2021-09-09:
Ok flag sounds good to me :)

Yatekii, 2021-09-09:
How about we enable it for defmt by default? There is really no reason for this in general RTT :/
I am kind of against adding this as a general default because it magically alters what the target does without the user expecting it. In the past I spent sooo many hours because things would not behave the same with and without a debugger :) In this case, one could argue both ways I guess as both kind of make it hard, but it's really a defmt problem and not an RTT one :)

derekdreery, 2021-09-09:
Ok maybe just make blocking opt-in for now since the problem should fix itself with defmt 0.3

derekdreery, 2021-09-09:
I'll try and do a PR for this if I get chance, but can't promise. I'm obviously not expecting anyone else to do one. :slightly_smiling_face:

Yatekii, 2021-09-09:
Sounds good to me :) I am kind of out of time where I do not work or sleep atm, but it will be better in at least one week and then I guess I can take a look too :)

ianrrees, 2022-02-25:
@derekdreery just wondering if you're still working on this?  I'd be happy to try making a PR to get this feature implemented - it would be quite useful.

ianrrees, 2022-02-25:
Somewhat related: https://github.com/mvirkkunen/rtt-target/issues/25

derekdreery, 2022-02-25:
> Somewhat related: mvirkkunen/rtt-target#25

I'm busy with my day job so go ahead and do it. Thanks very much.

ianrrees, 2022-03-03:
@derekdreery does that PR cover your use case?  If so, I suppose this can be closed out.

derekdreery, 2022-03-04:
Yeah looks really great, thanks for doing this!

closed 2022-03-04 by derekdreery