probe-rs/cargo-embed #272: Flashing of STM32L071KBTx with J-Link broken with 0.11.0 but works with 0.10.1

rnestler, 2021-11-09:
**Describe the bug**

When I try to flash my binary with `cargo embed` 0.11.0 my chip just hangs indefinitely. If I do the same with 0.10.1 it works. I also tried latest master which fails as well

**To Reproduce**
Steps to reproduce the behavior:
1. Flash with cargo embed 0.11.0
2. Broken

**Expected behavior**
It should work with 0.11.0 as well

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**
 - Linux (ArchLinux)

**Additional context**

J-Link:
```
J-Link: Firmware version: J-Link V9 compiled May  7 2021 16:26:12
J-Link: Hardware version: J-Link 9.40.0
```

rnestler, 2021-11-09:
I did a `git-bisect` in the https://github.com/probe-rs/probe-rs/ repository and used the cli tool there to do the download. The result was that https://github.com/probe-rs/probe-rs/commit/e7701ce3cfafab24f682351aa60decdf7a4584a7 is the breaking change. For me it looks like it enables double buffering for all chips.

rnestler, 2021-11-09:
Disabling double buffering with the following patch fixes it for me:
```diff
diff --git a/probe-rs/src/flashing/flasher.rs b/probe-rs/src/flashing/flasher.rs
index 7a86164d..e9cf08b4 100644
--- a/probe-rs/src/flashing/flasher.rs
+++ b/probe-rs/src/flashing/flasher.rs
@@ -102,7 +102,8 @@ impl<'session> Flasher<'session> {
     }

     pub(super) fn double_buffering_supported(&self) -> bool {
-        self.flash_algorithm.page_buffers.len() > 1
+        false
+        //self.flash_algorithm.page_buffers.len() > 1
     }

     fn load(&mut self) -> Result<(), FlashError> {
```

Yatekii, 2021-11-10:
What exactly happens without the patch? Does it just misflash without an error?

rnestler, 2021-11-10:
> What exactly happens without the patch? Does it just misflash without an error?

Depends. 0.11.0 did just misflash without an error, but on master I sometimes got errors during flashing as well:
```
Error failed to flash ~/projects/gfroerli/firmware/firmware/target/thumbv6m-none-eabi/release/examples/supply_monitor

             Caused by:
                 0: Error while flashing
                 1: The page write of the page at address 0x08000400 failed.
                 2: An error with the usage of the probe occured
                 3: Operation timed out
```

Yatekii, 2021-11-10:
Ok, so it actually seems that master fixed a problem where we did not report write issues properly. And it seems to me that the chip definition file might be wrong.

Yatekii, 2021-11-10:
Actually, this is really odd. Because it fails to access flash. This should not be dependant on double buffering. Double buffering does not use a different algorithm. All it does is write flash while it is filling a new RAM page. Could it be that this chip cannot access flash and RAM at the same time?

@adamgreig @korken89 @Tiwalun @Dirbaio any ideas what oddities the L0 could contain? The target description seems alright according to the ref man.

Dirbaio, 2021-11-10:
double-buffering writes to the second buffer while the algo is flashing the first buffer. I guess while it's flashing the bus is busy so the ram writes to the second buffer fail.

An easy fix would be to get `double_buffering_supported` from the target yamls, and set it to `false` for the chips that have trouble with it?

Alternatively, figure out why RAM writes fail, I have no idea how. Or retry them, but that could be dangerous for writes to registers etc...

rnestler, 2021-11-11:
> double-buffering writes to the second buffer while the algo is flashing the first buffer. I guess while it's flashing the bus is busy so the ram writes to the second buffer fail.

But this doesn't really explain the silent failures I saw, does it?

> Alternatively, figure out why RAM writes fail, I have no idea how.

Neither do I. I coukd take some time this evening to debug further, but I'm not familiar enough with the probe-rs codebase to really know where to look at.

Yatekii, 2021-11-11:
> > double-buffering writes to the second buffer while the algo is flashing the first buffer. I guess while it's flashing the bus is busy so the ram writes to the second buffer fail.
> 
> But this doesn't really explain the silent failures I saw, does it?

This is why I thought that we had a bug on 0.11.0 that didn't spot the memory transfer error, but it is fixed on master, which is why it errors properly.

Maybe I can drop by the coredump sometime? ;)