probe-rs/cargo-embed #285: Garbled output on UI from defmt target

TomDeRybel, 2021-12-09:
**Describe the bug**
The cargo-embed UI console displays garbled output for a target that uses defmt. 

An example UI console output:
`defmt`                                                                                                                                                                                                                                                       
`11:43:39.073 ~z`
`11:43:39.073 zzzzzzzz`
`11:43:39.381 zzzzzzzzzzzzzzzzzzzzzz z!z"z#z$z%z&z'z(z)z*z+z,z-.z/z0z1z2z3z4z5z6z7z8z9z:z;z<z=z>z?z@zAzBzCzDzEzFzGzHzIzJzKzLzMzNzOzPz`

I have attached the used Embed.toml file (extention modded for github), for reference. I have tried various options under [default.rtt] for channels, but to no avail.
[Embed_toml.txt](https://github.com/probe-rs/cargo-embed/files/7684453/Embed_toml.txt)

The relevant Cargo.toml lines for the embedded target (STM32H723ZG, a member of the newly-supported STM32H735 family) are:
defmt = "^0.3.0"
defmt-rtt = "^0.3.1"
panic-probe = { version = "^0.3.0", features = ["print-defmt"] }

I tried to do a local build of cargo-embed, with modified a Cargo.toml to use the newest defmt crate (and also modified a local probe-rs with the Cargo.toml modified likewise). This did not change the result.

Unfortunately, as this is a new chip family, I could not perform a test with probe-run, as the latter does not currently accept a custom YAML file for the processor family, and probe-rs lacks built-in support at present.

**To Reproduce**
1. Produce a firmware using the latest defmt that generates defmt data.
2. Run cargo-embed and observe the UI output.

**Expected behavior**
The defmt strings are not garbled.

**Desktop (please complete the following information):**
 - Linux (Ubuntu 20.04.3 LTS)

Yatekii, 2021-12-09:
I think this is because we did not update the rtt-decoder to 0.3 for cargo-embed yet.

probe-rs has built in support for loading custom YAML. But probe-rs is a library and as such does not make assumptions about it's environment, which is why you need to load it from code.

TomDeRybel, 2021-12-13:
Thank you for the clarification. I'll keep an eye out and will try again when an update comes out.

danielzfranklin, 2022-01-29:
@TomDeRybel Defmt works for me if I install `cargo-embed` from git HEAD (rtt-decoder was upgraded in https://github.com/probe-rs/cargo-embed/pull/283).

One gotcha is that `cargo-embed` specifies dependencies as git repos without specifying a specific version. As a result you can't (currently) build it without using the locked dependency versions, which cargo doesn't do by default.

So the full install command would be `cargo install --force --git https://github.com/probe-rs/cargo-embed.git --locked`

Yatekii, 2022-01-29:
Yes, this should indeed be fixed now :)

If the issue persists, please feel free to reopen :)

closed 2022-01-29 by Yatekii

TomDeRybel, 2022-01-30:
Thank you for the tip. I tried installing HEAD before, but that failed due to the gotcha you explained.
Happy to report that Defmt v3 debugging through cargo-embed works correctly now.