probe-rs/cargo-embed #325: Dependencies list is missing `libudev-dev`

rockboynton, 2022-03-13:
**Describe the bug**
The dependencies list for Ubuntu/Debian is missing `libudev-dev` from the `README` 

**To Reproduce**
Steps to reproduce the behavior:
1. On a fresh Ubuntu system, install Rust 
2. Run the dependency install command from the `README`
3. Run `cargo install cargo-embed`
4. See failure indicating missing `libudev` library

**Expected behavior**
I expect the cargo install command to work after installing dependencies

**Stacktrace**
```shell
...
   Compiling hidapi v1.3.4
   Compiling heck v0.3.3
   Compiling async-channel v1.6.1
error: failed to run custom build command for `hidapi v1.3.4`

Caused by:
  process didn't exit successfully: `/tmp/cargo-installTJ6YHF/release/build/hidapi-df8a287941c8cad9/build-script-build` (exit status: 101)  --- stdout
  cargo:rerun-if-env-changed=LIBUDEV_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=LIBUDEV_STATIC
  cargo:rerun-if-env-changed=LIBUDEV_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at 'Unable to find libudev: `"pkg-config" "--libs" "--cflags" "libudev"` did not exit successfully: exit status: 1
  error: could not find system library 'libudev' required by the 'hidapi' crate

  --- stderr
  Package libudev was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libudev.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libudev' found
  ', /home/rock/.cargo/registry/src/github.com-1ecc6299db9ec823/hidapi-1.3.4/build.rs:57:54
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-embed v0.12.0`, intermediate artifacts can be found at `/tmp/cargo-installTJ6YHF`

Caused by:
  build failed
```

**Desktop (please complete the following information):**
 - Windows 10 (WSL2 - Ubuntu)

amritabithi, 2022-05-06:
I can confirm this is still happening on the same configuration/steps and the same error messages as the original poster, with fresh install of Ubuntu 20 operating system.

Yatekii, 2022-05-14:
We gladly accept PRs fixing this :)

closed 2022-05-15 by bors[bot]

scrobot, 2022-05-23:
Hi, every1. 
I added additional instructions for Fedora-like distributive
https://github.com/probe-rs/cargo-embed/pull/341

commit 7eadfe6b (2022-05-15): Update the README to include `libudev-dev`

Fixes #325