probe-rs/cargo-embed #329: RTT logging to file fails on Windows

omion, 2022-04-18:
The RTT log file name is constructed using the time, but that has colons in it. Windows will not allow colons in filenames, so creating the file fails. I got the following error:

`Could not create log file ./logs\stm32_STM32H723ZGTx_2022-04-18T13:19:17.019966600-07:00_channel0.txt: The filename, directory name, or volume label syntax is incorrect. (os error 123)`

Windows 10 21H2, cargo-embed 0.12.0

Thanks!
- Reed

Tiwalun, 2022-04-19:
I think this is fixed in #290, but there hasn't been a release with that fix yet.

As a workaround, you can install the current version from master, which includes that fix:

```console
cargo install --git https://github.com/probe-rs/cargo-embed.git --locked
```

closed 2022-04-19 by Tiwalun

omion, 2022-04-19:
I tested out the master version, and it was fixed! I apologize for the noise.

(on a side note, I had to run your command without the `--locked` argument. The cargo.lock file in this repo specifies crossterm 0.23.0, which has issues with ctrl+c. Running it without `--locked` let Cargo use crossterm 0.23.1, which works)