probe-rs/cargo-embed #376: Turning off logging doesn't work

WaffleLapkin, 2022-09-06:
**Describe the bug**
You can't turn off logging via `Embed.toml`.

**To Reproduce**
Steps to reproduce the behavior:
1. Write the following to `Embed.toml`
   ```toml
   [default.general]
   log_level = "OFF"
   ```
3. Run `cargo embed`
4. Observe an error:
   ```text
          Error Failed to parse supplied configuration:
                unknown variant: found `OFF`, expected `one of `ERROR`, `WARN`, `INFO`, `DEBUG`, `TRACE`` for key "default.general.log_level" in Embed.toml TOML file
   ```

**Expected behavior**
`log_level = "OFF"` turns off logging, as suggested by [`default.toml`](https://github.com/probe-rs/cargo-embed/blob/c1218a540b815a5b46261bb9c7c66ad2ccd55e6d/src/config/default.toml#L40).

**Desktop:**
 - (Arch) Linux

Yatekii, 2022-09-06:
Yes, this is a mistake in the comment. To turn logging off, just remove 'log_level' :)

WaffleLapkin, 2022-09-06:
@Yatekii how would it work though? `log_level` is defined as a [`log::Level`](https://docs.rs/log/0.4.17/log/enum.Level.html) (not optional):
https://github.com/probe-rs/cargo-embed/blob/050a121dd1f66d9d097686cc2d4512e5e539220d/src/config/mod.rs#L70

Which can't ever be "off". 

Config is also merged with `default.toml`:
https://github.com/probe-rs/cargo-embed/blob/050a121dd1f66d9d097686cc2d4512e5e539220d/src/config/mod.rs#L107

... which has the `"WARN"` level by default:
https://github.com/probe-rs/cargo-embed/blob/050a121dd1f66d9d097686cc2d4512e5e539220d/src/config/default.toml#L41

Maybe [`log::LevelFilter`](https://docs.rs/log/0.4.17/log/enum.LevelFilter.html) was meant to be used instead?