probe-rs/cargo-embed #378: Install broken on macOS

badrbouslikhin, 2022-09-07:
**Describe the bug**
The install using `cargo install cargo-embed` seems broken.

**To Reproduce**
Steps to reproduce the behavior:
1. `cargo install cargo-embed`

**Stacktrace**
```
error[E0716]: temporary value dropped while borrowed
  --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-decoder-0.3.3/src/log/json_logger.rs:26:28
   |
26 |             let mut sink = io::stdout().lock();
   |                            ^^^^^^^^^^^^       - temporary value is freed at the end of this statement
   |                            |
   |                            creates a temporary which is freed while still in use
...
29 |             serde_json::to_writer(&mut sink, &create_json_frame(record, host_timestamp)).ok();
   |                                   --------- borrow later used here
   |
   = note: consider using a `let` binding to create a longer lived value

error[E0716]: temporary value dropped while borrowed
  --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-decoder-0.3.3/src/log/json_logger.rs:33:24
   |
33 |             let sink = io::stderr().lock();
   |                        ^^^^^^^^^^^^       - temporary value is freed at the end of this statement
   |                        |
   |                        creates a temporary which is freed while still in use
34 |             self.host_logger.print_host_record(record, sink);
   |                                                        ---- borrow later used here
   |
   = note: consider using a `let` binding to create a longer lived value

error[E0716]: temporary value dropped while borrowed
  --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-decoder-0.3.3/src/log/json_logger.rs:50:24
   |
50 |         let mut sink = io::stdout().lock();
   |                        ^^^^^^^^^^^^       - temporary value is freed at the end of this statement
   |                        |
   |                        creates a temporary which is freed while still in use
51 |         serde_json::to_writer(&mut sink, &SCHEMA_VERSION).ok();
   |                               --------- borrow later used here
   |
   = note: consider using a `let` binding to create a longer lived value

error[E0716]: temporary value dropped while borrowed
  --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-decoder-0.3.3/src/log/pretty_logger.rs:34:28
   |
34 |                 let sink = io::stdout().lock();
   |                            ^^^^^^^^^^^^       - temporary value is freed at the end of this statement
   |                            |
   |                            creates a temporary which is freed while still in use
...
38 |                     None => Self::print_println_record(record, sink),
   |                                                                ---- borrow later used here
   |
   = note: consider using a `let` binding to create a longer lived value

error[E0716]: temporary value dropped while borrowed
  --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-decoder-0.3.3/src/log/pretty_logger.rs:43:28
   |
43 |                 let sink = io::stderr().lock();
   |                            ^^^^^^^^^^^^       - temporary value is freed at the end of this statement
   |                            |
   |                            creates a temporary which is freed while still in use
44 |                 self.print_host_record(record, sink);
   |                                                ---- borrow later used here
   |
   = note: consider using a `let` binding to create a longer lived value

For more information about this error, try `rustc --explain E0716`.
error: could not compile `defmt-decoder` due to 5 previous errors
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-embed v0.13.0`, intermediate artifacts can be found at `/var/folders/6n/0wsjshxj7bg42ks97g69b9pc0000gn/T/cargo-installm3loUA`
```

**Desktop (please complete the following information):**
 - macOS Monterey (12.3.1)

Yatekii, 2022-09-07:
What Rust version is this with? Sounds like you use an outdated version of Rust, which leads to this :) The crate cannot be published if this never build in the first place, so I'd check that you use an up2date Rust version (rustup update).

badrbouslikhin, 2022-09-07:
@Yatekii it was v1.60.0
Upgrading to v1.63.0 fixes the issue. Thanks!

closed 2022-09-07 by badrbouslikhin