probe-rs/cargo-embed #389: `cargo embed -V` does not provide version number

sid2934, 2022-10-14:
**Describe the bug**
The  `cargo embed -V` command simply prints "cargo-embed" to the console with no indication of the version number running.
This is happening in Windows Terminal on Windows 11 Pro Build Version	10.0.22000 Build 22000
 

**To Reproduce**
Steps to reproduce the behavior:
1. Execute `cargo install cargo-embed` (Executed on October 14, 2022 likely installed version 0.13.0)
2. Run `cargo embed -V` or `cargo embed --version

**Expected behavior**
Expected result should have the version number included in the output

**Desktop**
Windows 11 Pro Build Version	10.0.22000 Build 22000

dbrgn, 2022-10-14:
Hehe, just ran across this issue as well, you were 1h faster in reporting 🙂

dbrgn, 2022-10-14:
The reason is that clap takes over handling of the version argument. I'll send a fix.

closed 2022-10-14 by bors[bot]

sid2934, 2022-10-14:
That was the fastest turn around for an issue I have ever filed. Thanks for the fix!

commit 75ab5c98 (2022-10-14): Merge #390

390: CLI: Fix missing version in -V/--version output r=Yatekii a=dbrgn

Clap overrides version handling by default, but we didn't pass any version to clap, so it's empty. Instead, disable version handling in clap and handle it ourselves.

Fixes #389.
PR #390 (2022-10-14): CLI: Fix missing version in -V/--version output
Clap overrides version handling by default, but we didn't pass any version to clap, so it's empty. Instead, disable version handling in clap and handle it ourselves.

Fixes #389.