probe-rs/cargo-embed #399: Compilation error in `goblin` dependency

Finomnis, 2022-10-27:
**Summary**
Trying to install `cargo-embed` via `cargo install cargo-embed` fails.

**To Reproduce**
System:
 - `rustc 1.64.0 (a55dd71d5 2022-09-19)`
 - `cargo 1.64.0 (387270bc7 2022-09-16)`
 - WSL2 running `Ubuntu 20.04.5 LTS`

Steps to reproduce the behavior:
1. `cargo install cargo-embed`

**Expected behavior**
`cargo-embed` gets installed.

**Actual behaviour**

The dependency `goblin` fails to compile:

```
    Updating crates.io index
  Installing cargo-embed v0.13.0
   Compiling proc-macro2 v1.0.47
   Compiling unicode-ident v1.0.5
   Compiling quote v1.0.21
   Compiling syn v1.0.103
   Compiling libc v0.2.137
   Compiling version_check v0.9.4
   Compiling serde_derive v1.0.147
   Compiling serde v1.0.147
   Compiling autocfg v1.1.0
   Compiling cfg-if v1.0.0
   Compiling memchr v2.5.0
   Compiling log v0.4.17
   Compiling cc v1.0.73
   Compiling once_cell v1.15.0
   Compiling pkg-config v0.3.26
   Compiling bitflags v1.3.2
   Compiling ryu v1.0.11
   Compiling itoa v1.0.4
   Compiling linked-hash-map v0.5.6
   Compiling rusb v0.9.1
   Compiling anyhow v1.0.66
   Compiling radium v0.7.0
   Compiling base64 v0.13.1
   Compiling thiserror v1.0.37
   Compiling serde_json v1.0.87
   Compiling byteorder v1.4.3
   Compiling parking_lot_core v0.9.4
   Compiling tap v1.0.1
   Compiling lazy_static v1.4.0
   Compiling fallible-iterator v0.2.0
   Compiling signal-hook v0.3.14
   Compiling stable_deref_trait v1.2.0
   Compiling regex-syntax v0.6.27
   Compiling io-lifetimes v0.7.4
   Compiling smallvec v1.10.0
   Compiling funty v2.0.0
   Compiling scopeguard v1.1.0
   Compiling rustix v0.35.12
   Compiling time-core v0.1.0
   Compiling termcolor v1.1.3
   Compiling semver v1.0.14
   Compiling camino v1.1.1
   Compiling static_assertions v1.1.0
   Compiling proc-macro-hack v0.5.19
   Compiling bitfield v0.14.0
   Compiling yansi v0.5.1
   Compiling ihex v3.0.0
   Compiling svg v0.10.0
   Compiling linux-raw-sys v0.0.46
   Compiling num_threads v0.1.6
   Compiling heck v0.4.0
   Compiling iana-time-zone v0.1.51
   Compiling os_str_bytes v6.3.0
   Compiling number_prefix v0.4.0
   Compiling paste v1.0.9
   Compiling managed v0.8.0
   Compiling utf8-width v0.1.6
   Compiling defmt-parser v0.3.1
   Compiling unsafe-libyaml v0.2.4
   Compiling plain v0.2.3
   Compiling textwrap v0.16.0
   Compiling either v1.8.0
   Compiling humantime v2.1.0
   Compiling difference v2.0.0
   Compiling inlinable_string v0.1.15
   Compiling scroll v0.10.2
   Compiling unicode-width v0.1.10
   Compiling strsim v0.10.0
   Compiling cassowary v0.3.0
   Compiling smawk v0.3.1
   Compiling unicode-segmentation v1.10.0
   Compiling dunce v1.0.3
   Compiling ahash v0.7.6
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling proc-macro2-diagnostics v0.9.1
   Compiling uncased v0.9.7
   Compiling figment v0.10.8
   Compiling indexmap v1.9.1
   Compiling num-traits v0.2.15
   Compiling lock_api v0.4.9
   Compiling num-integer v0.1.45
   Compiling yaml-rust v0.4.5
   Compiling libusb1-sys v0.6.4
   Compiling hidapi v1.4.2
   Compiling wyz v0.5.0
   Compiling gimli v0.26.2
   Compiling time-macros v0.2.5
   Compiling clap_lex v0.2.4
   Compiling byte-unit v4.0.14
   Compiling itertools v0.10.5
   Compiling getrandom v0.2.8
   Compiling atty v0.2.14
   Compiling signal-hook-registry v1.4.0
   Compiling time v0.1.44
   Compiling terminal_size v0.1.17
   Compiling time v0.3.16
   Compiling object v0.29.0
   Compiling aho-corasick v0.7.19
   Compiling bitvec v1.0.1
   Compiling colored v2.0.0
   Compiling console v0.15.2
   Compiling regex v1.6.0
   Compiling parking_lot v0.12.1
   Compiling terminal_size v0.2.1
   Compiling hashbrown v0.12.3
   Compiling indicatif v0.16.2
   Compiling sanitize-filename v0.4.0
   Compiling unicode-linebreak v0.1.4
   Compiling scroll_derive v0.11.0
   Compiling thiserror-impl v1.0.37
   Compiling enum-primitive-derive v0.2.2
   Compiling clap_derive v3.2.18
   Compiling git-version-macro v0.3.5
   Compiling pear_codegen v0.2.3
   Compiling scroll v0.11.0
   Compiling git-version v0.3.5
   Compiling pear v0.2.3
   Compiling clap v3.2.23
   Compiling textwrap v0.15.2
   Compiling jep106 v0.2.8
   Compiling serde_yaml v0.8.26
   Compiling bincode v1.3.3
   Compiling toml v0.5.9
   Compiling cargo-platform v0.1.2
   Compiling chrono v0.4.22
   Compiling serde_yaml v0.9.14
   Compiling probe-rs-target v0.13.0
   Compiling jaylink v0.3.0
   Compiling mio v0.8.5
   Compiling defmt-json-schema v0.1.0
   Compiling simplelog v0.12.0
   Compiling env_logger v0.9.1
   Compiling goblin v0.5.4
error[E0373]: closure may outlive the current function, but it borrows `foo`, which is owned by the current function
  --> /home/martin/.cargo/registry/src/github.com-1ecc6299db9ec823/goblin-0.5.4/src/lib.rs:13:25
   |
10 |     thread::scope(|scope| {
   |                    ----- has type `&'1 Scope<'1, '_>`
...
13 |             scope.spawn(|| {
   |                         ^^ may outlive borrowed value `foo`
...
16 |                     index, foo, s_ref
   |                            --- `foo` is borrowed here
   |
note: function requires argument type to outlive `'1`
  --> /home/martin/.cargo/registry/src/github.com-1ecc6299db9ec823/goblin-0.5.4/src/lib.rs:13:13
   |
13 | /             scope.spawn(|| {
14 | |                 println!(
15 | |                     "thread {} doing {} while having a borrow on {:?}",
16 | |                     index, foo, s_ref
17 | |                 )
18 | |             });
   | |______________^
help: to force the closure to take ownership of `foo` (and any other referenced variables), use the `move` keyword
   |
13 |             scope.spawn(move || {
   |                         ++++

error[E0373]: closure may outlive the current function, but it borrows `index`, which is owned by the current function
  --> /home/martin/.cargo/registry/src/github.com-1ecc6299db9ec823/goblin-0.5.4/src/lib.rs:13:25
   |
10 |     thread::scope(|scope| {
   |                    ----- has type `&'1 Scope<'1, '_>`
...
13 |             scope.spawn(|| {
   |                         ^^ may outlive borrowed value `index`
...
16 |                     index, foo, s_ref
   |                     ----- `index` is borrowed here
   |
note: function requires argument type to outlive `'1`
  --> /home/martin/.cargo/registry/src/github.com-1ecc6299db9ec823/goblin-0.5.4/src/lib.rs:13:13
   |
13 | /             scope.spawn(|| {
14 | |                 println!(
15 | |                     "thread {} doing {} while having a borrow on {:?}",
16 | |                     index, foo, s_ref
17 | |                 )
18 | |             });
   | |______________^
help: to force the closure to take ownership of `index` (and any other referenced variables), use the `move` keyword
   |
13 |             scope.spawn(move || {
   |                         ++++

error[E0373]: closure may outlive the current function, but it borrows `s_ref`, which is owned by the current function
  --> /home/martin/.cargo/registry/src/github.com-1ecc6299db9ec823/goblin-0.5.4/src/lib.rs:13:25
   |
10 |     thread::scope(|scope| {
   |                    ----- has type `&'1 Scope<'1, '_>`
...
13 |             scope.spawn(|| {
   |                         ^^ may outlive borrowed value `s_ref`
...
16 |                     index, foo, s_ref
   |                                 ----- `s_ref` is borrowed here
   |
note: function requires argument type to outlive `'1`
  --> /home/martin/.cargo/registry/src/github.com-1ecc6299db9ec823/goblin-0.5.4/src/lib.rs:13:13
   |
13 | /             scope.spawn(|| {
14 | |                 println!(
15 | |                     "thread {} doing {} while having a borrow on {:?}",
16 | |                     index, foo, s_ref
17 | |                 )
18 | |             });
   | |______________^
help: to force the closure to take ownership of `s_ref` (and any other referenced variables), use the `move` keyword
   |
13 |             scope.spawn(move || {
   |                         ++++

For more information about this error, try `rustc --explain E0373`.
error: could not compile `goblin` due to 3 previous errors
warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-embed v0.13.0`, intermediate artifacts can be found at `/tmp/cargo-installmJeI8x`
```

Yatekii, 2022-10-27:
Can you try using `--locked`?

Finomnis, 2022-10-27:
@Yatekii Seems like it installs without a problem with `--locked`.