probe-rs/cargo-embed #404: Host CPU usage increases over time

9names, 2022-11-20:
**Describe the bug**
When printing via defmt once per millisecond, CPU usage on the host machine grows.
Print output starts super smooth, but updates are clearly taking longer over time.
Eventually defmt can't print in time and the defmt stream is corrupted

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a target to print rapidly:
```rust
let mut counter = 0;
    loop {
        delay.delay_ms(1u32);
        debug!("{}", counter);
        counter += 1;
    }
```
2. 
```system
cargo embed --release
```
3. Watch CPU usage of cargo-embed grow by about 3% per second
4. No more messages printed once CPU usage hits 100%

**Expected behavior**
cargo-embed CPU usage should not be increasing over time

**Desktop (please complete the following information):**
 - Linux - Ubuntu 22.10

Yatekii, 2022-11-20:
Printing is very imperformant because of line wrapping and shenanigans like that. Everytime we update we recalculate the entire text we have so far. It is done very dumb atm and we were aware it will not be a solution forever.

Tiwalun, 2022-11-20:
I can reproduce it on a Mac, I see a similar growth in CPU usage.

Tiwalun, 2022-11-20:
The issue seems to be that all data received from the target is stored in binary form, and never thrown away.

This buffer gets larger and larger, and that means that parsing the buffer becomes slower and slower, since this complete buffer is parsed on every `render` call for the terminal interface.

Yatekii, 2022-11-20:
Yup, exactly. I am not sure how one fixes this properly with line wrapping. I guess we could do a chunking. Which only stores like 100 lines per chunk and loads new chunks when crossing the boundary.

9names, 2022-11-27:
The RTT case was pretty easy to work around in a naive way, since it's already split into lines inside `channel.rs`:
```rust
/// Maximum number of decoded line strings to keep buffered for UI display
/// TODO: make this a configuration setting
const MAXLINES:usize = 3000;
let lines = self.messages.len();
if lines > MAXLINES {
    self.messages = self.messages.split_off(lines - MAXLINES);
}
```
https://github.com/9names/cargo-embed/commit/833e725f7c390ed3556e92cdb04567574d751174
The clone every time we exceed MAXLINES is not the most efficient solution but it's practically noise compared to the rendering code.

We could decode defmt to lines inside `channel.rs` and then do a similar trick? It would simplify the logic inside `app::render()` a lot (since both cases are now just "draw these strings").

9names, 2022-11-27:
Ah, but cargo-embed relies on the fact that you maintain the buffer from the start of the debug session so it can flush it to disk on close?
Don't want to break that, will need to think about how to flush to disk the evicted entries.
Maybe split_off will work in our favour for this...

Tiwalun, 2022-11-27:
The biggest issue with `defmt` I found was that we perform the parsing step everythime. Just keeping the parsed messages should already help a lot, but that's a bit annoying because of the lifetime of the `demft` parser.

For saving to disk, I think just writing the log entries to disk regularly, maybe using a `BufWrite`, should help? I don't see a reason why the log files should only be written on close.

9names, 2022-11-27:
RTT has similar performance degradation, haven't measured the diff of RTT vs defmt but we really need to solve all 3 problems to resolve this issue completely.
The question is really: which to tackle first? I think getting all three formats to output Vec<String> or something similar would make the other problems easier to deal with. But: I'm still getting up to speed on the codebase so I'd appreciate any other suggestions as to how to approach this.

Tiwalun, 2022-11-27:
I don't know the codebase all that well myself, I'm more familiar with probe-rs. 

But maybe adding threads would be a good idea? So we could have a thread which handles input / output to the probe, so it would have the `probe-rs::Session`, as well as the whole defmt parsing. And then a separate thread handles the UI, and gets the messages from the I/O thread over one (or more) channels. The UI thread would then just keep the last n messages in a buffer, so that it can display them, and for some scrollback, if that is desired.

9names, 2022-11-28:
As a way of testing just how bad this would be, I moved the defmt and binle decoding from `render()` into `poll_rtt()`.
https://github.com/probe-rs/cargo-embed/compare/master...9names:cargo-embed:decode_in_channel
This was not as invasive as I thought it would be. Ownership of the defmt decoder hasn't changed (both functions are polled from the same location, decoder is passed by &) so it was actually rather trivial.
We do end up with an extra newline in the defmt output from somewhere, but since I just pushed the output through the old String output code without really looking too hard at the difference in the impl's it's not surprising that I broke it.

before
[image]
after
[image]