probe-rs/cargo-embed #407: detect symbols when a hex address is printed

ImUrX, 2022-11-23:
**Is your feature request related to a problem? Please describe.**
It would be very nice for backtraces being printed, so cargo-embed checks if it's a symbol and prints the name and location.
espflash does this as you can see

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ImUrX, 2022-11-23:
accidentaly sent it ;-;, i have edited the issue

Yatekii, 2022-11-24:
We gladly accept PRs :)