probe-rs/cargo-embed #409: Support using cargo-embed as a custom cargo runner

JoshMcguigan, 2022-11-25:
**Is your feature request related to a problem? Please describe.**

I'd like to be able to configure `cargo-embed` as a custom cargo runner, so `cargo run` can automatically use `cargo-embed`.

**Describe the solution you'd like**

In `.cargo/config.toml`:

```toml
runner = "cargo embed --maybe-some-new-arg-to-support-this-feature"
```

**Describe alternatives you've considered**

1. Run `cargo embed` rather than `cargo run`

matteocarnelos, 2022-11-29:
You can use [probe-run](https://github.com/knurling-rs/probe-run):
```
runner = "probe-run --chip <YOUR_CHIP>"
```

JoshMcguigan, 2022-11-29:
Hi, thanks for the suggestion! I considered probe-run, but I wanted to stick with cargo-embed since at the moment probe-run doesn't act as a gdb server. 

Although I've since run into #400 which is blocking gdb usage for me via cargo-embed anyway - so perhaps I'll give probe-run a go now.